nasa/osal #1243: Incorrect status reporting from OS_SocketSendTo_Impl

shuq0, 2022-04-01:
**Describe the bug**
The return value of `sendto` returns the number of characters sent on success and -1 on error. In `OS_SocketSendTo_Impl` the return of `sendto` is being assigned as os_status and returned. This results in a non-zero (`OS_SUCCESS`) status being reported back to the calling function which is interpreted as an error

**To Reproduce**
Steps to reproduce the behavior:
1. Put breakpoint on `OS_SocketSendTo_Impl`
2. Enable TO output using `cmdUtil`
3. Chaeck the value of `os_result`

**Expected behavior**
`os_result` should be OS_SUCCESS if the return value of `sendto != -1`. 

**Code snips**

```
osal/src/os/portable/os-impl-bsd-sockets.c

OS_SocketSendTo_Impl { ...

 os_result = sendto(impl->fd, buffer, buflen, MSG_DONTWAIT, sa, addrlen);
    if (os_result < 0)
    {
        OS_DEBUG("sendto: %s\n", strerror(errno));
        return OS_ERROR;
    }

    return os_result;
...
```

**Reporter Info**
Syed Huq, Kepler Communications.

skliper, 2022-04-04:
Not really clear what makes the current behavior "incorrect".  `OS_SocketSendTo` documents the current behavior:
https://github.com/nasa/osal/blob/88f72d788d33a12e7bf9d9f01c77346438730ffa/src/os/inc/osapi-sockets.h#L386-L403

> This results in a non-zero (OS_SUCCESS) status being reported back to the calling function which is interpreted as an error

The API is documented as returning actual bytes, so the calling function should handle positive return values.

shuq0, 2022-04-04:
Thanks for the clarification, fixed the calling function to account for this case.

closed 2022-04-04 by shuq0