nasa/osal #1244: Strict-type errors/exception when creating TimeBase callback on RTEMS during cFS init

pepepr08, 2022-04-01:
**Describe the bug**
A memory alignment exception occurs when running with RTEMS on GR740 board when compiling with `OMIT_DEPRECATED = true` and `MISSION_RESOURCEID_MODE = "STRICT"`

The exception occurs during cFS init, when starting the TimeBase callback task/thread:
https://github.com/nasa/osal/blob/main/src/os/rtems/src/os-impl-timebase.c#L392-L395

The callback thread `OS_TimeBase_CallbackThread` function argument is of type `osal_id_t`. 
https://github.com/nasa/osal/blob/main/src/os/shared/src/osapi-timebase.c#L381

RTEMS expects the callback function argument to be of type `rtems_task_argument`. The conversion from one type to another when starting the task causes the exception.

**To Reproduce**
Build for rtems5-gr740 with `OMIT_DEPRECATED = true` and `MISSION_RESOURCEID_MODE = "STRICT"`
Run cFS on gr740 board

**Expected behavior**
Successful cFS init.

**System observed on:**
 - Hardware: GR740
 - OS: RTEMS 5
 - Versions: Draco

**Reporter Info**
Jose F Martinez Pedraza / GSFC 582

skliper, 2022-07-20:
@acudmore - any opinions on this one?  Should probably get it resolved in Draco.

jphickey, 2022-07-20:
Just saw this (sorry I missed the mention from Apr 6) ... but yeah there should just be a wrapper function to deal with the different data type.  For example the POSIX implementation already has it, see OS_TimeBasePthreadEntry here: 

https://github.com/nasa/osal/blob/1de47bb10050b9bdae46974f24a34281b53e4949/src/os/posix/src/os-impl-timebase.c#L314-L321

I think `rtems_task_argument` was an int in 4.11 so a wrapper wasn't strictly needed but its definitely preferable (and more type safe, future proof) to use the entry point wrapper like POSIX has.

commit 942a1f67 (2022-08-04): Fix #1244, Add RTEMS timebase callback wrapper
commit e6a8537c (2022-08-04): Merge pull request #1276 from pepepr08/fix1244-rtemstimebase

Fix #1244, Add RTEMS timebase callback wrapper
closed 2022-08-04 by dzbaker