nasa/osal #1281: Rtems should show task names when typing `cpuuse`

thesamprice, 2022-08-22:
**Describe the bug**
typing `cpuuse` in console should show rtems names.

Add the following bits to `OS_TaskCreate_Impl` in rtems
```C
#include <rtems/score/threadimpl.h>
...
    ISR_lock_Context  lock_context_1;
    Thread_Control *the_thread;
...
    the_thread = _Thread_Get(impl->id, &lock_context_1 );
    _Thread_Set_name(the_thread, task->task_name);
    
```

**Expected behavior**
rtems should register its name, so it shows up when using rtems commands.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware microblaze
 - OS: rtems 6

**Additional context**
Better debug info in rtems

**Reporter Info**
Nasa

skliper, 2023-08-03:
@thesamprice - If you are using RTEMS 5 you can use pthread_setname_np which doesn't conflict with the object name.  I've got it working locally, see c3b8e26475ad9551d6c7a322843b4e196de09319.  It uses _Thread_Set_name under the hood, which utilizes a different structure than the object name itself.

Sample output:
```
-------------------------------------------------------------------------------
                              CPU USAGE BY THREAD
------------+----------------------------------------+---------------+---------
 ID         | NAME                                   | SECONDS       | PERCENT
------------+----------------------------------------+---------------+---------
 0x09010001 | IDLE                                   |      6.924443 |  86.332
 0x0a010001 | UI1                                    |      0.000000 |   0.000
 0x0a010002 | ntwk                                   |      0.002736 |   0.034
 0x0a010003 | ETH0                                   |      0.000000 |   0.000
 0x0a010004 | FTPa                                   |      0.000000 |   0.000
 0x0a010005 | FTPD                                   |      0.000000 |   0.000
 0x0a010007 | cFS                                    |      0.000000 |   0.000
 0x0a010008 |                                        |      0.000213 |   0.002
 0x0a010009 |                                        |      0.330194 |   4.115
 0x0a01000a | BSWP                                   |      0.003068 |   0.038
 0x0a01000b | BRDA                                   |      0.010960 |   0.136
 0x0a01000c | CFE_EVS                                |      0.001207 |   0.015
 0x0a01000d | shel                                   |      0.015485 |   0.193
 0x0a01000e | CFE_SB                                 |      0.001240 |   0.015
 0x0a01000f | CFE_ES                                 |      0.035894 |   0.447
 0x0a010010 | ES_BG_TASK                             |      0.001430 |   0.017
 0x0a010011 | CFE_TIME                               |      0.009163 |   0.114
 0x0a010012 | TIME_TONE_TASK                         |      0.005439 |   0.067
 0x0a010013 | TIME_1HZ_TASK                          |      0.003104 |   0.038
 0x0a010014 | CFE_TBL                                |      0.001444 |   0.018
 0x0a010015 | FIFO_LIB_ChildT                        |      0.011508 |   0.143
 0x0a010016 | CI_LAB_APP                             |      0.005689 |   0.070
 0x0a010017 | TO_LAB_APP                             |      0.012372 |   0.154
 0x0a010018 | SCH_LAB_APP                            |      0.003868 |   0.048
 0x0a010019 | HW                                     |      0.023436 |   0.291
 0x0a01001a | CF                                     |      0.002247 |   0.028
 0x0a01001b | DS                                     |      0.025332 |   0.315
 0x0a01001c | FM                                     |      0.015175 |   0.189
 0x0a01001d | FM_CHILD_TASK                          |      0.029992 |   0.373
 0x0a01001e | LC                                     |      0.021521 |   0.268
 0x0a01001f | SC                                     |      0.025404 |   0.316
 0x0a010020 | VENDI                                  |      0.019558 |   0.243
 0x0a010021 | DP                                     |      0.481416 |   5.996
------------+----------------------------------------+---------------+---------
 TIME SINCE LAST CPU USAGE RESET IN SECONDS:                          8.028488
-------------------------------------------------------------------------------
```

thesamprice, 2023-08-03:
@skliper Ill give that a try when I get a chance.

commit 14013523 (2023-09-13): Merge pull request #1407 from skliper:fix1281-set_rtems_task_name

Fix #1281, Set RTEMS task name for cpuuse
commit aadcb727 (2023-09-13): Fix #1281, Set RTEMS task name for cpuuse
PR #1407 (2023-09-13): Fix #1281, Set RTEMS task name for cpuuse
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #1281 

**Testing performed**
Ran on RTEMS 5.  Sample output:

```
------------------------------------------------------------------ [... cut]
closed 2023-09-13 by dzbaker