nasa/osal #1445: count-sem-test, bin-sem-test intermittently fail on POSIX as non-root user

ahatstat, 2024-01-31:
**Describe the bug**
count-sem-test and bin-sem-test unit tests fail intermittently on POSIX systems when running the tests as a non-elevated user when OSAL_CONFIG_DEBUG_PERMISSIVE_MODE is True. Both tests fail for the same reason. The tests spawn multiple tasks with different priority and assert that the highest priority task executes before the others.  When running the tests on Linux as a regular user it is not possible to set the thread priority. When OSAL_CONFIG_DEBUG_PERMISSIVE_MODE set to True any attempt to set the thread priority is ignored.  In this mode there is no guarantee which task executes first causing the tests to fail intermittently.  

**To Reproduce**
Build OSAL with tests for generic-linux on any Linux PC as described in the [quick start](https://github.com/nasa/osal#quick-start)
```
mkdir build_osal_test
cd build_osal_test
cmake -DENABLE_UNIT_TESTS=true -DOSAL_SYSTEM_BSPTYPE=generic-linux -DOSAL_CONFIG_DEBUG_PERMISSIVE_MODE=TRUE ..
make
make test
```
Run bin-sem-test or count-sem-test in a loop as a regular user.  It will eventually fail.  See the test outputs below for the failing case.
Note that running the tests in a loop as root does not seem to fail.  

**Expected behavior**
The tests should always pass.  

**Code snips**
[bin-sem-test.c](https://github.com/nasa/osal/blob/fed9ead356160fd3594ffadc02291b5cf54d1e5f/src/tests/bin-sem-test/bin-sem-test.c#L175C1-L195C44)
```
    /* Start 3 child tasks, give and confirm highest priority task increments, flush and confirm all three */
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[0], "Task_0", Test_BinSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0),
        OS_SUCCESS);
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[1], "Task_1", Test_BinSem_Task1, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(200), 0),
        OS_SUCCESS);
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[2], "Task_2", Test_BinSem_Task2, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(250), 0),
        OS_SUCCESS);
    OS_TaskDelay(100);
    UtAssert_UINT32_EQ(task_counter[0] + task_counter[1] + task_counter[2], 0);
    UtAssert_INT32_EQ(OS_BinSemGive(sem_id[0]), OS_SUCCESS);
    OS_TaskDelay(100);
    // Any of the following tests can fail
    UtAssert_UINT32_EQ(task_counter[0], 1);
    UtAssert_UINT32_EQ(task_counter[1] + task_counter[2], 0);**
    UtAssert_INT32_EQ(OS_BinSemFlush(sem_id[0]), OS_SUCCESS);
    OS_TaskDelay(100);
    UtAssert_UINT32_EQ(task_counter[0], 2);
    UtAssert_UINT32_EQ(task_counter[1], 1);
    UtAssert_UINT32_EQ(task_counter[2], 1);
```
[count-sem-test.c](https://github.com/nasa/osal/blob/fed9ead356160fd3594ffadc02291b5cf54d1e5f/src/tests/count-sem-test/count-sem-test.c#L177C1-L192C62)
```
    /* Start 3 child tasks, give and confirm highest priority task increments, flush and confirm all three */
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[0], "Task_0", Test_CountSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0),
        OS_SUCCESS);
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[1], "Task_1", Test_CountSem_Task1, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(200), 0),
        OS_SUCCESS);
    UtAssert_INT32_EQ(
        OS_TaskCreate(&task_id[2], "Task_2", Test_CountSem_Task2, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(250), 0),
        OS_SUCCESS);
    OS_TaskDelay(100);
    UtAssert_UINT32_EQ(task_counter[0] + task_counter[1] + task_counter[2], 0);
    UtAssert_INT32_EQ(OS_CountSemGive(sem_id[0]), OS_SUCCESS);
    OS_TaskDelay(100);
    // These tests can fail
    UtAssert_UINT32_EQ(task_counter[0], 1);
    UtAssert_UINT32_EQ(task_counter[1] + task_counter[2], 0);
```

**System observed on:**
 - PC
 - OS: Ubuntu 22.04
 - Versions: [OSAL Equuleus-rc1+dev2] (https://github.com/nasa/osal/commit/fed9ead356160fd3594ffadc02291b5cf54d1e5f)

**Additional context**

**Reporter Info**
Andrew Hatstat, Geost

ahatstat, 2024-01-31:
Here is the output of `count-sem-test` when it fails
```
[BEGIN] UNIT TEST

[BEGIN] 01 SETUP
OS_Posix_GetSchedulerParams():202:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():202:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():388:Selected policy 2 for RT tasks, root task = 99
OS_Posix_TaskAPI_Impl_Init():405:Could not setschedparam in main thread: Operation not permitted (1)
[  END] No test cases

[BEGIN] 01 Test_CountSem
[ PASS] 01.001 count-sem-test.c:86 - OS_CountSemGetInfo(OS_OBJECT_ID_UNDEFINED, &sem_prop) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.002 count-sem-test.c:87 - OS_CountSemGive(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.003 count-sem-test.c:88 - OS_CountSemTake(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.004 count-sem-test.c:89 - OS_CountSemTimedWait(OS_OBJECT_ID_UNDEFINED, 0) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.005 count-sem-test.c:90 - OS_CountSemDelete(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.006 count-sem-test.c:93 - OS_CountSemCreate(NULL, "Test_Sem", 0, 0) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.007 count-sem-test.c:94 - OS_CountSemCreate(&sem_id[0], NULL, 0, 0) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.008 count-sem-test.c:95 - OS_CountSemGetInfo(sem_id[0], NULL) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.009 count-sem-test.c:96 - OS_CountSemGetIdByName(NULL, "Test_Sem") (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.010 count-sem-test.c:97 - OS_CountSemGetIdByName(&sem_id[0], NULL) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.011 count-sem-test.c:101 - OS_CountSemCreate(&sem_id[0], long_name, 0, 0) (-13) == OS_ERR_NAME_TOO_LONG (-13)
[ PASS] 01.012 count-sem-test.c:102 - OS_CountSemGetIdByName(&sem_id[0], long_name) (-13) == OS_ERR_NAME_TOO_LONG (-13)
[ PASS] 01.013 count-sem-test.c:105 - OS_CountSemCreate(&sem_id[0], "Test_Sem", 0, 0) (0) == OS_SUCCESS (0)
[ PASS] 01.014 count-sem-test.c:106 - OS_CountSemCreate(&sem_id[1], "Test_Sem", 0, 0) (-15) == OS_ERR_NAME_TAKEN (-15)

**BUG** OS_CountSemCreate():90:check '(sem_id) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_CountSemCreate():91:check '(sem_name) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_CountSemGetInfo():230:check '(count_prop) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_CountSemGetIdByName():209:check '(sem_id) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_CountSemGetIdByName():210:check '(sem_name) != NULL' FAILED --> OS_INVALID_POINTER

[ PASS] 01.015 count-sem-test.c:109 - OS_CountSemCreate(&sem_id[1], "Test_Sem_Nonzero", 2, 0) (0) == OS_SUCCESS (0)
[ PASS] 01.016 count-sem-test.c:117 - OS_CountSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.017 count-sem-test.c:118 - sem_prop.value (0) == 0 (0)
[ PASS] 01.018 count-sem-test.c:119 - OS_CountSemGetInfo(sem_id[1], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.019 count-sem-test.c:120 - sem_prop.value (2) == 2 (2)
[ PASS] 01.020 count-sem-test.c:124 - OS_CountSemTimedWait(sem_id[0], 0) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.021 count-sem-test.c:125 - OS_CountSemTimedWait(sem_id[0], 1) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.022 count-sem-test.c:128 - OS_CountSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.023 count-sem-test.c:129 - OS_CountSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.024 count-sem-test.c:130 - OS_CountSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.025 count-sem-test.c:133 - OS_CountSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.026 count-sem-test.c:134 - sem_prop.value (3) == 3 (3)
[ PASS] 01.027 count-sem-test.c:136 - OS_CountSemTake(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.028 count-sem-test.c:137 - OS_CountSemTimedWait(sem_id[0], 0) (0) == OS_SUCCESS (0)
[ PASS] 01.029 count-sem-test.c:138 - OS_CountSemTimedWait(sem_id[0], 1) (0) == OS_SUCCESS (0)
[ PASS] 01.030 count-sem-test.c:139 - OS_CountSemTimedWait(sem_id[0], 0) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.031 count-sem-test.c:142 - OS_CountSemTimedWait(sem_id[1], 1) (0) == OS_SUCCESS (0)
[ PASS] 01.032 count-sem-test.c:143 - OS_CountSemTimedWait(sem_id[1], 0) (0) == OS_SUCCESS (0)
[ PASS] 01.033 count-sem-test.c:148 - OS_CountSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.034 count-sem-test.c:149 - sem_prop.value (0) == 0 (0)
[ PASS] 01.035 count-sem-test.c:150 - OS_CountSemGetInfo(sem_id[1], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.036 count-sem-test.c:151 - sem_prop.value (0) == 0 (0)
[ PASS] 01.037 count-sem-test.c:160 - OS_TaskCreate(&task_id[0], "Task_0", Test_CountSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.038 count-sem-test.c:35 - OS_CountSemGetIdByName(&sem_id, "Test_Sem") (0) == OS_SUCCESS (0)
[ PASS] 01.039 count-sem-test.c:164 - OS_TaskDelete(task_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.040 count-sem-test.c:165 - task_counter[0] (0) == 0 (0)
[ PASS] 01.041 count-sem-test.c:169 - OS_CountSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.042 count-sem-test.c:170 - sem_prop.value (0) == 0 (0)
[ PASS] 01.043 count-sem-test.c:178 - OS_TaskCreate(&task_id[0], "Task_0", Test_CountSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.044 count-sem-test.c:181 - OS_TaskCreate(&task_id[1], "Task_1", Test_CountSem_Task1, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(200), 0) (0) == OS_SUCCESS (0)
[... cut ...]
[ PASS] 01.079 count-sem-test.c:207 - task_counter[0] + task_counter[1] + task_counter[2] (9) == test_val (9)
[ PASS] 01.080 count-sem-test.c:210 - OS_CountSemDelete(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.081 count-sem-test.c:211 - OS_CountSemDelete(sem_id[1]) (0) == OS_SUCCESS (0)
[ INFO]                        ABORT::0     WARN::0     FLOW::0     DEBUG::0     N/A::0
[  END] 01 Test_CountSem        TOTAL::81    PASS::79    FAIL::2     MIR::0     TSF::0     TTF::0

[BEGIN] 02 Cleanup
[  END] No test cases
COMPLETE: 1 tests Segment(s) executed
```

ahatstat, 2024-01-31:
here is the output of `bin-sem-test` when it fails
```
[BEGIN] 01 Test_BinSem
[ PASS] 01.001 bin-sem-test.c:86 - OS_BinSemGetInfo(OS_OBJECT_ID_UNDEFINED, &sem_prop) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.002 bin-sem-test.c:87 - OS_BinSemFlush(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.003 bin-sem-test.c:88 - OS_BinSemGive(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.004 bin-sem-test.c:89 - OS_BinSemTake(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.005 bin-sem-test.c:90 - OS_BinSemTimedWait(OS_OBJECT_ID_UNDEFINED, 0) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.006 bin-sem-test.c:91 - OS_BinSemDelete(OS_OBJECT_ID_UNDEFINED) (-16) == OS_ERR_INVALID_ID (-16)
[ PASS] 01.007 bin-sem-test.c:94 - OS_BinSemCreate(NULL, "Test_Sem", 0, 0) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.008 bin-sem-test.c:95 - OS_BinSemCreate(&sem_id[0], NULL, 0, 0) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.009 bin-sem-test.c:96 - OS_BinSemGetInfo(sem_id[0], NULL) (-2) == OS_INVALID_POINTER (-2)
[ PASS] 01.010 bin-sem-test.c:97 - OS_BinSemGetIdByName(NULL, "Test_Sem") (-2) == OS_INVALID_POINTER (-2)

**BUG** OS_BinSemCreate():98:check '(sem_id) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_BinSemCreate():99:check '(sem_name) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_BinSemGetInfo():259:check '(bin_prop) != NULL' FAILED --> OS_INVALID_POINTER

**BUG** OS_BinSemGetIdByName():238:check '(sem_id) != NULL' FAILED --> OS_INVALID_POINTER

[ PASS] 01.011 bin-sem-test.c:98 - OS_BinSemGetIdByName(&sem_id[0], NULL) (-2) == OS_INVALID_POINTER (-2)

**BUG** OS_BinSemGetIdByName():239:check '(sem_name) != NULL' FAILED --> OS_INVALID_POINTER

[ PASS] 01.012 bin-sem-test.c:102 - OS_BinSemCreate(&sem_id[0], long_name, 0, 0) (-13) == OS_ERR_NAME_TOO_LONG (-13)
[ PASS] 01.013 bin-sem-test.c:103 - OS_BinSemGetIdByName(&sem_id[0], long_name) (-13) == OS_ERR_NAME_TOO_LONG (-13)
[ PASS] 01.014 bin-sem-test.c:106 - OS_BinSemCreate(&sem_id[0], "Test_Sem", 0, 0) (0) == OS_SUCCESS (0)
[ PASS] 01.015 bin-sem-test.c:107 - OS_BinSemCreate(&sem_id[1], "Test_Sem", 0, 0) (-15) == OS_ERR_NAME_TAKEN (-15)
[ PASS] 01.016 bin-sem-test.c:110 - OS_BinSemCreate(&sem_id[1], "Test_Sem_Nonzero", 1, 0) (0) == OS_SUCCESS (0)
[ PASS] 01.017 bin-sem-test.c:118 - OS_BinSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.018 bin-sem-test.c:119 - sem_prop.value (0) == 0 (0)
[ PASS] 01.019 bin-sem-test.c:120 - OS_BinSemGetInfo(sem_id[1], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.020 bin-sem-test.c:121 - sem_prop.value (1) == 1 (1)
[ PASS] 01.021 bin-sem-test.c:125 - OS_BinSemTimedWait(sem_id[0], 0) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.022 bin-sem-test.c:126 - OS_BinSemTimedWait(sem_id[0], 1) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.023 bin-sem-test.c:129 - OS_BinSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.024 bin-sem-test.c:131 - OS_BinSemTake(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.025 bin-sem-test.c:132 - OS_BinSemTimedWait(sem_id[0], 0) (-7) == OS_SEM_TIMEOUT (-7)
[ PASS] 01.026 bin-sem-test.c:135 - OS_BinSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.027 bin-sem-test.c:136 - OS_BinSemTimedWait(sem_id[0], 0) (0) == OS_SUCCESS (0)
[ PASS] 01.028 bin-sem-test.c:137 - OS_BinSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.029 bin-sem-test.c:138 - OS_BinSemTimedWait(sem_id[0], 1) (0) == OS_SUCCESS (0)
[ PASS] 01.030 bin-sem-test.c:141 - OS_BinSemTimedWait(sem_id[1], 0) (0) == OS_SUCCESS (0)
[ PASS] 01.031 bin-sem-test.c:146 - OS_BinSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.032 bin-sem-test.c:147 - sem_prop.value (0) == 0 (0)
[ PASS] 01.033 bin-sem-test.c:148 - OS_BinSemGetInfo(sem_id[1], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.034 bin-sem-test.c:149 - sem_prop.value (0) == 0 (0)
[ PASS] 01.035 bin-sem-test.c:158 - OS_TaskCreate(&task_id[0], "Task_0", Test_BinSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.036 bin-sem-test.c:35 - OS_BinSemGetIdByName(&sem_id, "Test_Sem") (0) == OS_SUCCESS (0)
[ PASS] 01.037 bin-sem-test.c:162 - OS_TaskDelete(task_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.038 bin-sem-test.c:163 - task_counter[0] (0) == 0 (0)
[ PASS] 01.039 bin-sem-test.c:167 - OS_BinSemGetInfo(sem_id[0], &sem_prop) (0) == OS_SUCCESS (0)
[ PASS] 01.040 bin-sem-test.c:168 - sem_prop.value (0) == 0 (0)
[ PASS] 01.041 bin-sem-test.c:176 - OS_TaskCreate(&task_id[0], "Task_0", Test_BinSem_Task0, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(100), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.042 bin-sem-test.c:179 - OS_TaskCreate(&task_id[1], "Task_1", Test_BinSem_Task1, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(200), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.043 bin-sem-test.c:49 - OS_BinSemGetIdByName(&sem_id, "Test_Sem") (0) == OS_SUCCESS (0)
[ PASS] 01.044 bin-sem-test.c:182 - OS_TaskCreate(&task_id[2], "Task_2", Test_BinSem_Task2, NULL, TASK_STACK_SIZE, OSAL_PRIORITY_C(250), 0) (0) == OS_SUCCESS (0)
[ PASS] 01.045 bin-sem-test.c:35 - OS_BinSemGetIdByName(&sem_id, "Test_Sem") (0) == OS_SUCCESS (0)
[ PASS] 01.046 bin-sem-test.c:63 - OS_BinSemGetIdByName(&sem_id, "Test_Sem") (0) == OS_SUCCESS (0)
[ PASS] 01.047 bin-sem-test.c:186 - task_counter[0] + task_counter[1] + task_counter[2] (0) == 0 (0)
[ PASS] 01.048 bin-sem-test.c:187 - OS_BinSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.049 bin-sem-test.c:53 - OS_BinSemTake(sem_id) (0) == OS_SUCCESS (0)
[ FAIL] 01.050 bin-sem-test.c:189 - task_counter[0] (0) == 1 (1)
[... cut ...]
[ PASS] 01.074 bin-sem-test.c:208 - OS_BinSemGive(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.075 bin-sem-test.c:210 - task_counter[0] + task_counter[1] + task_counter[2] (9) == test_val (9)
[ PASS] 01.076 bin-sem-test.c:213 - OS_BinSemDelete(sem_id[0]) (0) == OS_SUCCESS (0)
[ PASS] 01.077 bin-sem-test.c:214 - OS_BinSemDelete(sem_id[1]) (0) == OS_SUCCESS (0)
[ INFO]                        ABORT::0     WARN::0     FLOW::0     DEBUG::0     N/A::0
[  END] 01 Test_BinSem          TOTAL::77    PASS::73    FAIL::4     MIR::0     TSF::0     TTF::0
```

skliper, 2024-01-31:
Looks like it's failing due to the non-root scheduling.  Likely needs more relaxed asserts in the case where priorities aren't strictly enforced.

commit b16cb4bb (2025-07-15): Fix #1445, add conditional around priority sem test

The multi-thread sem tests (count + bin) have a test that relies on the
priority taking effect, that the highest priority thread will get the
sem first.  However when running as non-root and in permissive mode, the
priority is not applied and thus this becomes a race condtion.

This adds a test and skips the test if priority is not enabled