nasa/osal #1447: Add generic RTEMS tar file system support

skliper, 2024-02-02:
**Is your feature request related to a problem? Please describe.**
Current PC RTEMS workflow creates a fat32 disk image and QEMU command line drive option for runtime access. See:
https://github.com/nasa/cFS/blob/main/.github/scripts/rtems-test.mk

Flight systems often utilize flat nonvol memory for boot/startup which can be done using RTEMS support for linking a tared filesystem with the executable and populating the root file system from the tar file at startup.

**Describe the solution you'd like**
Add generic support for RTEMS tar filesystem

**Describe alternatives you've considered**
EEFS is an alternative simple file system, adding RTEMS tar filesystem support doesn't preclude using other options.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

closed 2024-04-01 by dzbaker

DavidEdell, 2024-06-21:
How do you utilize this new TARFS support?

I can enable it by setting RTEMS_INCLUDE_TARFS, however that results in a link error because the tar file itself is not generated and linked in.  I don't see anything committed that would generate the tar file. 
 Is there another piece that has not been pushed/merged?  

I'm ultimately looking at getting the latest cfe version running on one of our Leon3 boards.

thanks,
-David

skliper, 2024-06-21:
We are using it on a Leon3.  Along with https://github.com/nasa/PSP/pull/376, we added a "tarfs" target to do the last stage.  So the process is to build for the leon3 with the following settings:
```
set(RTEMS_DYNAMIC_LOAD      TRUE)
set(RTEMS_INCLUDE_TARFS     TRUE)
set(RTEMS_NO_CMDLINE        TRUE)
```
Then we just do a `cmake TARGET=tarfs prep`.  It's kludgy at this point since we just needed something that worked for us, but see https://github.com/nasa/cFE/compare/main...skliper:cFE:EXAMPLE-tarfs.  Note I just threw this into the cFE sample_defs to share, it relies on gaisler tools, etc so it won't just work out of the box.

PR #1350 (2024-04-01): Fix #1349 #1447, Break up pc-rtems to support generic configuration and add tarfs support
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #1349 
- Fix #1447

Splits up bsp_start.c so features can be optionally included: cmd [... cut]