nasa/osal #1472: Incorrect function parameter types in os-impl-no-select.c

samuel-krain, 2024-07-16:
**Describe the bug**
The file `.../portable/os-impl-no-select` has two functions, `OS_SelectSingle_Impl()` and `OS_SelectMultiple_Impl()`. These both have the final parameter `int32 msecs`, which does not match the header definitions in `os-shared-select.h` where the final parameter is instead `OS_time_t abs_timeout`. 

As a result, there is a compiler error when the file `os-impl-no-select` is compiled.

**Note:** files are only compiled when using RTEMS OS with libnetworking library disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone OSAL repo into empty directory
2. Add the rtems6 toolchain file from CFE Repo
3. Modify toolchain file:

    1.  Modify path to relevant RTEMS6 compiler tools
    2.  Set `CMAKE_SYSTEM_VERSION` to `6`
    3.  Select `pc-rtems` BSP
    4.  Select i386/pc686 architecture

5. Add the `RTEMS.cmake` platform module from the PSP repo in `./Platform`
6. Add the following lines before `project()` in top-level CMakeLists.txt:

```cmake
list(APPEND CMAKE_MODULE_PATH "$ENV{HOME}/<path-to-project>")
include(Platform/RTEMS)
```

7. Run the following commands:

```bash
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=debug \
    -DENABLE_UNIT_TESTS=true \
    -DOSAL_CONFIG_DEBUG_PERMISSIVE_MODE=true \
    -DOSAL_CONFIG_INCLUDE_NETWORK=FALSE \
    -DCMAKE_TOOLCHAIN_FILE="path/to/toolchain-pc686-rtems.cmake" \
    ../osal
make
```

8. See following error:

```console
...
[ 18%] Building C object osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/src/os-impl-loader.c.o
[ 19%] Building C object osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/__/portable/os-impl-posix-dl-symtab.c.o
[ 20%] Building C object osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/__/portable/os-impl-no-network.c.o
[ 21%] Building C object osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/__/portable/os-impl-no-sockets.c.o
[ 21%] Building C object osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/__/portable/os-impl-no-select.c.o
/<path>/osal/src/os/portable/os-impl-no-select.c:60:7: error: conflicting types for 'OS_SelectSingle_Impl'; have 'int32(const OS_object_token_t *, uint32 *, int32)' {aka 'int(const struct OS_object_token *, unsigned int *, int)'}
   60 | int32 OS_SelectSingle_Impl(const OS_object_token_t *token, uint32 *SelectFlags, int32 msecs)
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /<path>/osal/src/os/portable/os-impl-no-select.c:32:
/<path>/osal/src/os/shared/inc/os-shared-select.h:52:7: note: previous declaration of 'OS_SelectSingle_Impl' with type 'int32(const OS_object_token_t *, uint32 *, OS_time_t)' {aka 'int(const struct OS_object_token *, unsigned int *, OS_time_t)'}
   52 | int32 OS_SelectSingle_Impl(const OS_object_token_t *token, uint32 *SelectFlags, OS_time_t abs_timeout);
      |       ^~~~~~~~~~~~~~~~~~~~
/<path>/osal/src/os/portable/os-impl-no-select.c:71:7: error: conflicting types for 'OS_SelectMultiple_Impl'; have 'int32(OS_FdSet *, OS_FdSet *, int32)' {aka 'int(OS_FdSet *, OS_FdSet *, int)'}
   71 | int32 OS_SelectMultiple_Impl(OS_FdSet *ReadSet, OS_FdSet *WriteSet, int32 msecs)
      |       ^~~~~~~~~~~~~~~~~~~~~~
/<path>/osal/src/os/shared/inc/os-shared-select.h:75:7: note: previous declaration of 'OS_SelectMultiple_Impl' with type 'int32(OS_FdSet *, OS_FdSet *, OS_time_t)' {aka 'int(OS_FdSet *, OS_FdSet *, OS_time_t)'}
   75 | int32 OS_SelectMultiple_Impl(OS_FdSet *ReadSet, OS_FdSet *WriteSet, OS_time_t abs_timeout);
      |       ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/build.make:426: osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/__/portable/os-impl-no-select.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:386: osal/rtems_impl/CMakeFiles/osal_rtems_impl.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

**Expected behavior**
Compilation without errors.

**Code snips**
os-impl-no-select.c:

```C
/****************************************************************************************
                                    INCLUDE FILES
 ***************************************************************************************/

#include <osapi.h>
#include "os-shared-select.h"

...

/*----------------------------------------------------------------
 *
 *  Purpose: Implemented per internal OSAL API
 *           See prototype for argument/return detail
 *
 *-----------------------------------------------------------------*/
int32 OS_SelectSingle_Impl(const OS_object_token_t *token, uint32 *SelectFlags, int32 msecs)
{
    return OS_ERR_NOT_IMPLEMENTED;
}

/*----------------------------------------------------------------
 *
 *  Purpose: Implemented per internal OSAL API
 *           See prototype for argument/return detail
 *
 *-----------------------------------------------------------------*/
int32 OS_SelectMultiple_Impl(OS_FdSet *ReadSet, OS_FdSet *WriteSet, int32 msecs)
{
    return OS_ERR_NOT_IMPLEMENTED;
}
```
os-shared-select.h:
```c
...
int32 OS_SelectSingle_Impl(const OS_object_token_t *token, uint32 *SelectFlags, OS_time_t abs_timeout);
...
int32 OS_SelectMultiple_Impl(OS_FdSet *ReadSet, OS_FdSet *WriteSet, OS_time_t abs_timeout);
...
```

**Additional Context**

OSAL does not appear to be compatible with libbsp library which replaces libnetworking in RTEMS6. Therefore, networking is more likely to be disabled.

**System observed on:**
 - Hardware:
     
    | Component | Value |
    |--------|--------|
    | processor | 12th Gen Intel(R) Core(TM) i7-12700K |
    | memory   | 64GiB System memory |
    | display     | GA104 [GeForce RTX 3060 Ti Lite Hash Rate] |  

 - OS: Ubuntu 22.04.4 LTS (kernel 6.5.0)
 - Versions: OSAL commit d6412df in main (latest as of 07-16-24)

**Reporter Info**
Samuel Krain

skliper, 2024-07-16:
@dzbaker - is it possible to prioritize this one along w/ a CI review as to how this wasn't caught (and confirm there isn't a systematic issue)?

jphickey, 2025-03-13:
Note - internal GSFC issue has been written to address the follow on work to check for missing CI/unit-tests that should have caught this.

After merging #1495 I would recommend that this issue be closed.

skliper, 2025-03-13:
@jphickey - does it really have to be an internal issue?  Or, how do those that are following the open source follow internal open issues?

closed 2025-03-27 by dzbaker