nasa/osal #1488: printf improvements - snprintf, int return value

CDKnightNASA, 2024-12-12:
**Is your feature request related to a problem? Please describe.**
There are multiple Gateway VSM applications using snprintf and it would be good to wrap in OSAL to provide coverage test stubs and consistency with the OS_printf wrapper.

Also, currently, OS_printf() is defined to return void. Most (all?) OS implementations of printf returns the number of bytes written (or a negative number on an error.)

**Describe the solution you'd like**
Change OS_printf() to return int32.

Add OS_snprintf() (and OS_vsnprintf()?) wrappers.

**Describe alternatives you've considered**
Leave it up to app developers to write their own wrappers, as needed.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2024-12-12:
Note also that the `UT_DefaultHandler_OS_printf()` has special handling of a simple `"%s"` format string, for performance reasons (?), yet it calls `vsnprintf()` above that logic...Why the special logic?

jphickey, 2024-12-12:
In general for functions defined in ISO C / C99 we do not provide an abstraction -- there is not a clear benefit for doing so, as all compliant C language implementations are required to provide these functions.  There was an exception for printf / OS_printf because there was a need to make it buffered/asynchronous -- on many embedded systems it is unbuffered and thus blocks the caller while characters are written to the console, sometimes at 9600 bps or less.

While I concur that the ISO specification for printf has it returning an `int` -- most programmers don't check the return value and even if they did, there isn't much to do about a failure to printf.

Regarding stubs and the ability to stub out the function -- this often causes confusion because the output is often needed.  That is, its one of those cases like memset/memcpy/strcpy/etc where the stub would have to invoke the real thing to preserve dependent logic.

CDKnightNASA, 2024-12-13:
Joe, is there an easy way to stub out standard functions like snprintf()? I'm guessing that a wrapper would be required.

There are some other issues with the doxygen documentation and whatnot with OS_printf, so I may make a PR anyways.

Lastly, I stumbled across the `restrict` keyword. We don't use it anywhere but it might be worthwhile to start using it to give hints to the compiler for optimization. It is C99. https://en.cppreference.com/w/c/language/restrict Would love to hear your thoughts on it at a CCB.

skliper, 2024-12-13:
> Joe, is there an easy way to stub out standard functions like snprintf()? I'm guessing that a wrapper would be required.
 
I'm curious as to why you'd want to do this.  Are you trying to force snprintf failures?  If there's a submodule that really needs to override built in functions you can do it similar to how it's done by OSAL or PSP but I'd avoid it unless there's a really good reason.  See src/unit-test-coverage/ut-stubs that provides OCS_xxx versions of functions.

skliper, 2024-12-13:
PSP example of snprintf override:
https://github.com/nasa/PSP/blob/cdc73e5f6564f678207bec69cb75d456795ce91c/unit-test-coverage/ut-stubs/src/PCS_stdio_stubs.c#L189

CDKnightNASA, 2024-12-16:
> I'm curious as to why you'd want to do this. Are you trying to force snprintf failures? If there's a submodule that really needs to override built in functions you can do it similar to how it's done [...]

We check the return of snprintf in our code and, in order to perform coverage tests, need to stub out snprintf. It wouldn't surprise me if there would be someone who would also want to check the return value of printf to confirm success or handle an error (assuming OS_printf returned an int) as well, FYI.

CDKnightNASA, 2024-12-17:
> PSP example of snprintf override: https://github.com/nasa/PSP/blob/cdc73e5f6564f678207bec69cb75d456795ce91c/unit-test-coverage/ut-stubs/src/PCS_stdio_stubs.c#L189

Thanks Jake, and looks like https://github.com/nasa/PSP/blob/cdc73e5f6564f678207bec69cb75d456795ce91c/unit-test-coverage/ut-stubs/override_inc/stdio.h is the trick to get unit/coverage tests to use this stub. We'll look at using this in our coverage test code. Given that even PSP uses a stub for snprintf, any interest in wrapping it in OSAL or is the minimum criteria that OSAL adds some functionality on top?

Thoughts on my other concerns about OS_printf()? There's also some documentation cleanup that would be good for OS_printf.

jphickey, 2024-12-17:
Yes - for coverage stubs of C99 functions, the trick is to use the "override_inc" that maps it to the OCS/PCS version (that stands for OSAL coverage stub, and PSP coverage stub, respectively -- but that doesn't matter, the main thing is that its a different name).  This effectively replaces all invocations of `snprintf` with `OCS_snprintf` instead.

Important to note that the "override_inc" is by no means trying to be a general substitute/replacement for the C99 headers.  They have the same names but only expose/define symbols that the FSW actually uses or otherwise needs in order to compile and run.  The values for symbols probably wouldn't work if you tried to use them as intended by C99.  They are strictly for coverage, but also limited to coverage based on the way PSP/OSAL uses these functions.  A different user, with a different use-case, may require different items from the override header.  Example - in the manual there is an errno that printf can set that has to do with wide char support - `EILSEQ`.  But our override for stdio.h/errno.h does not define this, because we don't use it (nor any other wide-char stuff) in OSAL or PSP, despite it being part of C99 (I think).

To make a general-purpose set of override includes would basically creep into implementing an entire set of C99-compliant headers and API calls, and nobody is trying to do that!  We only want to compile OSAL/PSP for the purpose of getting coverage.  Likewise, if a developer of some other app decides they need to stub out `snprintf` for some reason, that's fine -- they can do so using the patterns in OSAL as a guide.  But they would have to clone and own - this is intentional, because we do not, under any circumstances, make anything even resembling a claim to provide general purpose C99 stubs in UT Assert.  But we should caution against this -- it seems tempting/convenient to "lie" about a return value to get the branch you want, but this often has side effects when the return value doesn't match the operation that was actually done.  All our FSW was organized in a way that allows this to happen, but C99 stuff is kinda fundamental and stuff breaks if it doesn't behave exactly as advertised, both in FSW and UT.

dzbaker, 2025-01-02:
CCB 2 January 2025: Closing as is (see discussion). Recommend following pattern in PSP and OSAL (overrides).

closed 2025-01-02 by dzbaker