nasa/osal #1513: gethostid() and gethostname() not supported by lwIP network

mafoste3, 2025-08-05:
**Describe the bug**
The lwIP network stack, compatible with RTEMS, does not support either gethostid() and gethostname() calls. If these OSAL networking modules are included then the build will fail with undefined references to the calls.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware
 - OS: [e.g. Linux 4.4]
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable

joelsherrill, 2025-08-05:
Please report this to RTEMS.org. It is ok for OSAL to work around the current lacking but it would be better long term if these can be added. Sometimes, the functionality is there but not promoted out via a standard API. 

There is an lwip channel in Discord for chatting about this. But an issue at gitlab.rtems.org will be needed.

mafoste3, 2025-08-09:
RTEMS Issue created: [gethostid() and gethostname() not supported by lwIP network](https://gitlab.rtems.org/rtems/pkg/rtems-lwip/-/issues/8#top)

opticron, 2025-08-28:
@mafoste3 I have written an implementation for rtems-lwip in https://gitlab.rtems.org/rtems/pkg/rtems-lwip/-/merge_requests/28 if you wouldn't mind giving it a try.